AssessingSolar/solarstations #9: Generate additional metadata

AdamRJensen, 2022-02-18:
It would be nice to provide additional metadata for the stations, including:
* Climate classification ([Köppen Geiger](https://en.wikipedia.org/wiki/K%C3%B6ppen_climate_classification))
* Annual irradiance (just GHI or all three components?)
* Continent (useful for sorting the stationlisting and statistics)

It would be optimal to automatically generate this metadata. The annual irradiance could come from a provider with global coverage, e.g., [NASA Power](https://power.larc.nasa.gov/). The continent could be determined based on latitude/longitude.

Additional things to consider:
* In this [paper](http://les.edu.uy/FRS/extras/Gueymard_RA_Difusa_SE2015.pdf) by Gueymard only four climate classifications were used. Are these more relevant for solar energy applications?

kanderso-nrel, 2022-02-18:
You may also be interested in PV Climate Zones: https://pvtools.lbl.gov/pv-climate-stressors and https://github.com/toddkarin/pvcz

AdamRJensen, 2022-02-18:
Thanks @kanderso-nrel. Is PV Climate Zone actually something that is used?

I noticed that the first link also shows Köppen Geiger climate zones, so perhaps I can get some inspiration for how to program that.

kanderso-nrel, 2022-02-18:
It's a relatively new development (especially compared w/ something like KG), but yes it is getting used: https://scholar.google.com/scholar?cites=7515697662620840254&as_sdt=4005&sciodt=0,6&hl=en

kandersolar, 2024-02-05:
This package has high-resolution KG classification lookup: https://pypi.org/project/kgcPy/

closed 2024-10-29 by AdamRJensen