AssessingSolar/solarstations #17: Add KACARE stations

AdamRJensen, 2022-03-29:
SolarAnywhere benchmarking used a number of high-quality stations from KACARE (King Abdullah City for Atomic and Renewable Energy): https://www.solaranywhere.com/resources/data-validation/

[image]

AdamRJensen, 2024-01-30:
Email sent to the Solar Energy authors on 30-01-2024 to enquire about which stations are still active.

The [API](https://www.energy.gov.sa:8443/index.html) provided only gives metadata.

IoannisSifnaios, 2024-03-05:
More material [here](https://www.mdpi.com/2071-1050/10/4/1129) and the actual map of stations with photos is [here](https://rratlas.energy.gov.sa/RRMMDataPortal/MapTool).

Some info on instruments is [here](https://rratlas.energy.gov.sa/RRMMPublicPortal/sites/default/files/Solar%20Resource%20Data%20Documentation.htm).

The network looks active.

AdamRJensen, 2024-03-15:
What is your "looks active" comment based on? Have you seen that you are able to get recent data or?

IoannisSifnaios, 2024-03-15:
No, I did not get any data (it required to make an account, etc. so I did not look into that). I said that it "looks active" because they have grayed out some stations in their atlas, showing that they are not in operation. So I assumed that the rest are in operation... of course, that makes sense as long as they have an updated website (which I am not sure of)! I am also ok to change them to "?" if you think it makes more sense.

AdamRJensen, 2024-03-15:
Question mark'em and I will merge it!

IoannisSifnaios, 2024-03-15:
> Question mark'em and I will merge it!

Done!

PR #154 (2024-03-15): Add KACARE stations
closes #17
closed 2024-03-15 by AdamRJensen