AssessingSolar/solarstations #38: Regional Radiation Centres (RRC)

AdamRJensen, 2022-08-07:
It should be investigated if not all of the [Regional Radiation Centres (RRC)](https://community.wmo.int/activity-areas/imop/radiation-centres) designated by the WMO make continuous solar surface irradiance measurements and can be added to the list. Some of the RRCs are already included in the list, e.g., Lindenberg, Norrköping, and Carpentras.

Description from the WMO:
> A Regional Radiation Centre is a centre designated by a Regional Association to serve as a centre for interregional comparisons of radiation instruments within the Region and to maintain the standard instruments necessary for this purpose.

**Region I (Africa)**
- [ ] Cairo (Egypt)
- [ ] Khartoum (Sudan)
- [ ] Kinshasa (Dem. Rep. of the Congo)
- [ ] Lagos (Nigeria)
- [x] Tamanrasset (Algeria) (BSRN)
- [ ] Tunis (Tunisia)

**Region II (Asia)**
- [ ] Pune (India)
- [ ] Tokyo (Japan)

**Region III (South America)**
- [ ] Buenos Aires (Argentina)
- [ ] Lima (Peru)
- [ ] Santiago (Chile)

**Region IV (North and Central America)**
- [ ] Toronto (Canada)
- [ ] Boulder (United States)
- [ ] Mexico City (Mexico)

**Region V (South-West Pacific)**
- [ ] Melbourne (Australia)

**Region VI (Europe)**
- [ ] Budapest (Hungary)
- [ ] Davos (Switzerland)
- [ ] St. Petersburg (Russian Federation)
- [x] Norrköping (Sweden) (BSRN)
- [x] Trappes/Carpentras (France) (BSRN)
- [ ] Uccle (Belgium)
- [x] Lindenberg (Germany) (BSRN)

IoannisSifnaios, 2024-02-06:
[Melbourne](https://community.wmo.int/en/activity-areas/imop/Regional_Instrument_Centres/RIC_Australia)

[Cairo](https://community.wmo.int/en/activity-areas/imop/Regional_Instrument_Centres/ric-egypt)

IoannisSifnaios, 2024-02-09:
For Sudan, a number of stations exist measuring only GHI, described [here](https://www.sciencedirect.com/science/article/pii/S0196890499001235). I could not find information about stations measuring more components.

IoannisSifnaios, 2024-02-12:
In Central Africa, there seems to be a station network called [FAOCLIM 2](https://www.fao.org/land-water/land/land-governance/land-resources-planning-toolbox/category/details/en/c/1028000/) that has solar irradiance data, but only measuring GHI. A list of stations is described [here](https://www.sciencedirect.com/science/article/pii/S0169809523001084).

AdamRJensen, 2024-02-12:
@IoannisSifnaios Thanks for leaving these bread crumb trails even though we aren't including them. It will be highly useful in case anybody ever starts searching for Africa specific data.

IoannisSifnaios, 2024-02-14:
In Tunisia, the [National Institute of Meteorology (NIM)](https://www.meteo.tn/en) operates stations that measure the GHI. A number of articles also mention stations that measure GHI (e.g., [1](https://ietresearch.onlinelibrary.wiley.com/doi/epdf/10.1049/iet-rpg.2017.0118), [2](https://journals.indexcopernicus.com/api/file/viewByFileId/482798), [3](https://link.springer.com/article/10.1007/s00339-018-1551-3)).

The [GEBA network](https://geba.ethz.ch/data-retrieval/metadata.html) mentions a station in Tunis that operated from 1964-2008, but only measuring GHI.

IoannisSifnaios, 2024-02-15:
Some info on Pune [here ](https://www.imdpune.gov.in/SI/radiation_lab_new.html)and [here](https://imdpune.gov.in/index.php)

IoannisSifnaios, 2024-02-24:
In Peru, there is a network measuring only GHI. It is described [here](https://www.senamhi.gob.pe/load/file/01401SENA-9.pdf).

In a document from SENAMHI (National Meteorological and Hydrological Service) there is a photo of a tracker, but no more information could be found about it. The photo is in this [report](https://library.wmo.int/viewer/53114/download?file=P1_54_Vargas_AWSinPeru.pdf&type=pdf).

IoannisSifnaios, 2024-02-24:
Information on Toronto [here](https://gawsis.meteoswiss.ch/GAWSIS/index.html#/search/station/stationReportDetails/0-20008-0-TOT).

IoannisSifnaios, 2024-02-28:
@AdamRJensen in Russia, there is a netwrok of stations described [here](http://wrdc.mgo.rssi.ru/wwwrootnew/publ/WRDC_issue_2022_4.pdf) that only measure GHI and DHI.

The RRC station in St. Petersburg is also measuring GHI and DHI, as described [here](http://wrdc.mgo.rssi.ru/).

There is this network called "The World Network" that is coordinated by Russia and collaborated with stations in many countries (described [here](http://wrdc.mgo.rssi.ru/)). However, it seems that they don't measure DNI and they don't give information on how they measure the diffuse.

I am inclined not to include them, however, we have very little coverage in Russia. Thoughts?

AdamRJensen, 2024-02-28:
With that information i suggest we also don't include them

closed 2024-02-28 by IoannisSifnaios