AssessingSolar/solarstations #39: Reginal Test Centers (RTC) - USA

AdamRJensen, 2022-08-07:
The [Regional Test Centers](https://rtc.sandia.gov/) consists of five sites for testing PV outdoors for different climates.

- [ ] **Albuquerque, NM**
> The NM RTC is co-located with Sandia’s Photovoltaic Systems Evaluation Laboratory, which is a multiple-user facility that conducts research on emerging PV technologies and supports the detailed characterization and performance analysis of modules and components.

- [ ] **Golden, CO**
> The CO RTC is co-located with the National Renewable Energy Laboratory’ Outdoor Test Facility.

- [ ] **Calumet, MI**
> The Michigan Regional Test Center (MI RTC) is located in Calumet, Michigan adjacent to the Advanced Power Systems Center at Michigan Technological University.

- [ ] **Cocoa, FL**
> The FL RTC is located on a 10-acre site owned by the Florida Solar Energy Center in Cocoa, FL, where the climate is consistently hot and humid.

- [ ] **Henderson, NV**
> Located less than 20 miles south of Las Vegas, this site is defined by high irradiance, high heat and limited precipitation. Site Partners: the University of Nevada, Las Vegas and the Southern Nevada Water Authority (SNWA).

[image]

AdamRJensen, 2022-08-07:
Photo of the RTC in Cocoa
[image]

kandersolar, 2024-02-06:
@AdamRJensen my understanding is that the RTC irradiance datasets are not public, nor are there any plans to make them public.  A few datasets seem to have "leaked out" through collaborators, but to my knowledge none of them include the base irradiance components.  

I see "not available" is an option for [data availability](https://solarstations.org/station_requirements.html#station-metadata).  Does that mean we should still add the RTC stations to the list?

AdamRJensen, 2024-02-06:
Interesting! My impression was that if you contacted them, they would be willing to share data (I forget who I talked to, think it was a woman at Sandia).

But yes everything gets added! 😄 For now you can just put "Not available"

PR #124 (2024-02-06): Add US DOE RTC stations
Closes #39
closed 2024-02-06 by AdamRJensen