AssessingSolar/solarstations #56: Crosscheck stations from Solargis validation report

AdamRJensen, 2022-10-05:
https://solargis2-web-assets.s3.eu-west-1.amazonaws.com/public/doc/Validation-Report_Global-Solar-Atlas-2.0_WB-ESMAP_Nov2019-1.pdf

AdamRJensen, 2024-02-05:
Site name Country GHI DNI Latitude [°] Longitude [°] Elevation [m a.s.l.] Source Climate zone Recently updated 
A Coruna Spain x x 43.3672 -8.4194 58 AEMET 3 No
Adam Oman x x 22.2072 57.5230 250 OPWP 2 Yes 
Aggeneys South Africa x x -29.2945 18.8155 789 Eskom 2 Yes
Agoufu Mali x - 15.3445 -1.4791 290 - 2 Yes 
American Samoa  ASM  x  x  -14.2474  -170.5644  42  NOAA ESRL  1  No 
Amman Jordan x - 32.0247 35.8789 1041 - 3 No 
Armazones Chile x - -24.6346 -70.2426 2576 Ministerio de 2 Yes 
Athens Greece x - 37.9718 23.7183 107 IDMP 3 Yes
BacNinh Vietnam x x 21.2015 106.0629 60 ESMAP 3 Yes 
Bahawalpur Pakistan x x 29.3254 71.8188 123 ESMAP 2 Yes
Bamba Mali x - 17.0990 -1.4018 272 - 2 Yes
Banizoumbou Niger x - 13.5311 2.6613 211 - 2 Yes 
Barcelona Spain x - 41.3858 2.1169 125 SOLARFLUX 3 No 
Bergen Norway x - 60.3838 5.3319 45 IDMP Univ. of 3 Yes

The above stations are missing (only checked the first page of validation sites)

IoannisSifnaios, 2024-03-11:
@AdamRJensen I suggest we only investigate stations used to validate DNI (page 52 onwards), which is approximately half of the total stations included in this report.

If you approve all pending pull requests (in order to be absolutely sure which stations are already included), I am willing to take up the task. From a quick look at the DNI stations, I think that most of them are already included.

Thoughts?

AdamRJensen, 2024-03-11:
I agree with this approach. The table of missing stations above is also only of the DNI stations - I fear we are missing quite a few stations...

IoannisSifnaios, 2024-03-19:
The following stations measuring DNI are not included in the current list. 

Site name, Country, Lat, Lon, Elev, Source
- [ ] Adam, Oman, 22.2072, 57.5230, 250, OPWP
- [ ] Manah, Oman, 22.6031, 57.6672, 345, OPWP
- [ ] Aggeneys, South Africa, -29.2945, 18.8155, 789, Eskom
- [ ] Helios, South Africa, -30.5011, 19.5607, 905, Eskom
- [ ] BacNinh, Vietnam, 21.2015, 106.0629, 60, ESMAP
- [ ] Central  Highlands, Vietnam, 12.7534, 107.8763, 290, ESMAP
- [ ] Bahawalpur, Pakistan, 29.3254, 71.8188, 123, ESMAP
- [ ] Hyderabad, Pakistan, 25.4134, 68.2595, 63, ESMAP
- [ ] Islamabad, Pakistan, 33.6419, 72.9838, 558, ESMAP
- [ ] Khuzdar, Pakistan, 27.8178, 66.6294, 1254, ESMAP
- [ ] Karachi, Pakistan, 24.9334, 67.1116, 45, ESMAP
- [ ] Lahore, Pakistan, 31.6946, 74.2441, 207, ESMAP
- [ ] Multan, Pakistan, 30.1654, 71.4978, 123, ESMAP
- [ ] Quetta, Pakistan, 30.2708, 66.9398, 1586, ESMAP
- [ ] Bukit Kototabang, Indonesia, -0.2019, 100.3181, 864, GAW
- [ ] Caceres, Spain, 39.4722, -6.3394, 405, AEMET
- [ ] Cordoba, Spain, 37.8444, -4.8506, 91, AEMET
- [ ] Lleida, Spain, 41.6258, 0.5950, 192, AEMET
- [ ] Madrid, Spain, 40.4528, -3.7242, 664, AEMET
- [ ] Malaga, Spain, 36.7192, -4.4803, 60, AEMET
- [ ] Murcia, Spain, 38.0028, -1.1694, 62, AEMET
- [ ] Oviedo, Spain, 43.3536, -5.8733, 336, AEMET
- [ ] Palma, Spain, 39.5667, 2.7439, 4, AEMET
- [ ] San Bartolome Tirajana, Canary Isl., 27.7581, -15.5756, 50, AEMET
- [ ] San Sebastian, Spain, 43.3075, -2.0394, 252, AEMET
- [ ] Soria, Spain, 41.7667, -2.4667, 1082, AEMET
- [ ] A Coruna, Spain, 43.3672, -8.4194, 58, AEMET
- [ ] Valladolid, Spain, 41.6500, -4.7667, 735, AEMET
- [ ] Chilanga, Malawi, -15.6798, 34.9723, 767, ESMAP
- [ ] Chileka, Malawi, -15.6798, 34.9723, 767, ESMAP
- [ ] Kasungu, Malawi, -13.0153, 33.4685, 1065, ESMAP
- [ ] Mzuzu, Malawi, -11.4199, 33.9953, 1285, ESMAP
- [ ] Dar es Salaam, Tanzania, -6.7811, 39.2039, 93, ESMAP
- [ ] Dicheto, Ethiopia, 11.9156, 41.5511, 431, IFC
- [ ] Edinburg, Texas, USA, 26.3059, -98.1716, 45, NREL
- [ ] Lafayette, Louisiana, USA, 30.2050, -92.3979, 5, NREL
- [ ] Watkins, USA, 39.7568, -104.6202, 1674, NREL
- [ ] Egbert, Canada, 44.2300, -79.7800, 233, SACC
- [ ] Feni, Bangladesh, 22.8003, 91.3582, 15, ESMAP
- [ ] Freiburg, Germany, 47.9792, 7.8311, 275, IDMP
- [ ] Weihenstephan, Germany, 48.4000, 11.7000, 472
- [ ] Gan, Maldives, -0.6906, 73.1501, 2, ESMAP
- [ ] Hanimaadhoo, Maldives, 6.7464, 73.1686, 2, ESMAP
- [ ] Hulhulé, Maldives, 4.1927, 73.5281, 2, ESMAP
- [ ] Kadhdhoo, Maldives, 1.8583, 73.5197, 2, ESMAP
- [ ] Geneve, Switzerland, 46.2003, 6.1316, 420, CUEPE
- [ ] Hradec Kralove, Czech Republic, 50.1830, 15.8330, 236, CHMU
- [ ] Hrazdan, Armenia, 40.5116, 44.8230, 1845, ESMAP
- [ ] Masrik, Armenia, 40.2077, 45.7645, 1944, ESMAP
- [ ] Talin, Armenia, 40.3860, 43.8927, 1641, ESMAP
- [ ] Yerevan, Armenia, 40.1887, 44.3976, 946, ESMAP
- [ ] Hurso, Ethiopia, 9.6136, 41.6385, 1110, IFC
- [ ] Longe, Zambia, -14.8397, 24.9319, 1167, ESMAP
- [ ] Lusaka, Zambia, -15.3946, 28.3372, 1262, ESMAP
- [ ] Misamfu, Zambia, -10.1726, 31.2231, 1382, ESMAP
- [ ] Mochipapa, Zambia, -16.8382, 27.0703, 1282, ESMAP
- [ ] Mount Makulu, Zambia, -15.5483, 28.2482, 1224, ESMAP
- [ ] Mutanda, Zambia, -12.4236, 26.2153, 1317, ESMAP
- [ ] Vaulx un Velin, France, 45.7786, 4.9225, 170, IDMP

IoannisSifnaios, 2024-03-22:
Regarding the ESKOM network in South Africa (information was found [here](https://sterg.sun.ac.za/wp-content/uploads/2018/07/Meyer-ST-051.pdf)):

Eskom makes use of pyrheliometers mounted on trackers, with only some of their stations equipped with photoelectric pyranometers. Eskom data was reported to be of low quality due to a lack of regular cleaning. This is due to the remote locations of their stations in the Northern Cape.

Some information on ESKOM is [here](https://citeseerx.ist.psu.edu/document?repid=rep1&type=pdf&doi=e30bd3e44ab5d5d5c5f16032b4a062d2c2651f2a). It sounds like the stations had initially RSIs but they were replaced with trackers and pyrheliometers. They are probably Tier 2 stations, using shadowrings for measuring diffuse (based on a photo in the report).

IoannisSifnaios, 2024-03-23:
Information on Bukit Kototabang [here](https://gawsis.meteoswiss.ch/GAWSIS/#/search/station/stationReportDetails/0-20008-0-BKT).

IoannisSifnaios, 2024-03-23:
Information on Freiburg [here](https://www.sciencedirect.com/science/article/pii/S0038092X24000136).

Although the station's coordinates are not mentioned, it is clearly stated that G;B;D are measured with a tracker

IoannisSifnaios, 2024-03-24:
Information on some more stations in Europe was found [here ](https://hal.science/hal-00361364/document)and [here](https://asr.copernicus.org/articles/17/143/2020/).

IoannisSifnaios, 2024-03-25:
The Geneva station mentioned in this list has been moved since 2010, so the mentioned coordinates in the SolarGIS report are not accurate. Information about this are provided [here](https://access.archive-ouverte.unige.ch/access/metadata/a4d8798a-87a1-46cc-a3d3-7fbaa70a93ff/download).

The new station is located in Carouge, on the Battelle campus (latitude: 46.176°N, longitude: 6.139°E, altitude: 432m), and has GHI and DNI on a tracker (Tier 2). It belongs to the University of Geneva.

IoannisSifnaios, 2024-04-02:
Information on the Hradec Kralove station [here ](https://gawsis.meteoswiss.ch/GAWSIS/#/search/station/stationReportDetails/0-20008-0-HKR) and [here](http://www.cmes.cz/sites/default/files/MZ_2007_4_slunoměry.pdf).

PR #185 (2024-04-05): Add stations from SolarGIS report
Closes #56
closed 2024-04-05 by AdamRJensen