AssessingSolar/solarstations #63: Add Shagaya station in Kuwait

AdamRJensen, 2022-12-26:
Details:

Shagaya,,,Kuwait,29.2099,47.0603,241,,KISR,,,https://www.kisr.edu.kw/en/facilities/33/details/,,,,

AdamRJensen, 2024-02-02:
https://www.sciencedirect.com/science/article/pii/S0038092X17311404

> The Kuwait Institute for Scientific Research (KISR) is developing and maintaining a solar measurement network in support of achieving a sustainable energy mix for Kuwait ([Al-Rasheedi et al., 2014](https://www.sciencedirect.com/science/article/pii/S0038092X17311404#b0005)). Two of the five operational stations have reported 1-min irradiance data since 2012, using Kipp & Zonen thermopile radiometers (CHP1 pyrheliometers and CMP21 pyranometers) and a photodiode-based RSI (4G model). The latter instrument was developed by Reichert GmbH in collaboration with Solar Millennium AG, based on the LI-COR LI-200 silicon detector and the original design by [Michalsky et al. (1986)](https://www.sciencedirect.com/science/article/pii/S0038092X17311404#b0125). In Kuwait, this instrument was installed, calibrated, and software-corrected by [CSP](https://www.sciencedirect.com/topics/engineering/concentrated-solar-power) Services GmbH (hereafter, CSPS), using proprietary corrections developed (or adopted) by CSPS.

> A detail of the experimental setup is shown in [Fig. 1](https://www.sciencedirect.com/science/article/pii/S0038092X17311404#f0005). The station’s design, instrumentation, calibration, and maintenance program follow the best practices ([Sengupta et al., 2017](https://www.sciencedirect.com/science/article/pii/S0038092X17311404#b0160)). It is stressed that the cleaning and maintenance of all radiometers are routinely done (and documented photographically) seven days per week at each of these stations, barring [adverse conditions](https://www.sciencedirect.com/topics/engineering/adverse-condition). Such adverse conditions might be related to (i) the local weather (dust storms, fog, or rain); (ii) temporary station shutdowns; (iii) the unavailability of maintenance personnel; or (iv) exceptional circumstances, such as military exercises. [Fig. 2](https://www.sciencedirect.com/science/article/pii/S0038092X17311404#f0010) provides the number and description of major causes of disruption of the daily maintenance schedule at Shagaya during the 24-month experimental period discussed here. Overall, that station was serviced daily more than 90% of the time. This remarkable achievement under remote desert situations is a unique feature of the KISR network. Using the extensive daily photographic record of before/after cleaning pictures, [Fig. 3](https://www.sciencedirect.com/science/article/pii/S0038092X17311404#f0015) shows examples of soiling on a pyranometer dome. Experience showed that rain—a rare event in Kuwait—can actually increase the dust soiling effect, which is somewhat counterintuitive.

[image]
Fig. 1.

IoannisSifnaios, 2024-02-02:
https://proceedings.ises.org/paper/eurosun2014/eurosun2014-0137-AlRasheedi.pdf

PR #103 (2024-02-05): Kuwait stations
Closes #63
closed 2024-02-05 by AdamRJensen