AssessingSolar/solarstations #96: Fix legend in the intro map

AdamRJensen, 2024-01-30:
The legend in the intro map is currently not showing correctly.

Also, consider changing the legend title to "Station status" and making three bullets - Active, inactive, and unknown.

[image]

IoannisSifnaios, 2024-02-05:
@AdamRJensen I think it is a good idea to add a third status called "unknown". E.g., I could not find info if the Saudi Arabian or the Chilean network is still in operation, and by default they would be categorized as "active". However, that might require going over the stations again to make sure that they are in the correct category...

AdamRJensen, 2024-02-20:
@kandersolar What do you think of copying the way they've done the legend [here](https://nbviewer.org/gist/talbertc-usgs/18f8901fc98f109f2b71156cf3ac81cd)? From my research it seems like the best option.. its just a lot of ugly html

kandersolar, 2024-02-20:
I think the issue with the circle markers not appearing in the legend is that they are from an older version of font-awesome (v4?) and the website is now using v6, which doesn't have them.  The pydata-sphinx-theme vendors font-awesome, and I'm not sure if it's possible to configure what version it uses.  Want me to hack together a fix, using the same markers that the map is using instead of the font-awesome symbols?

For the irradiance map color scale, I would love to hear a good solution.  For PV Atlas I have a bunch of javascript that I'd be quite happy to ditch in favor of something better: https://github.com/kandersolar/pvatlas/blob/main/website/source/_static/map-widget.js#L115-L165

[image]

AdamRJensen, 2024-03-15:
> I think the issue with the circle markers not appearing in the legend is that they are from an older version of font-awesome (v4?) and the website is now using v6, which doesn't have them. The [...]

This would be great @kandersolar!

IoannisSifnaios, 2024-03-15:
I suggest we change the colors, though. To me, it would make sense as follows:
- active: green
- inactive: red
- unknown: blue or grey

Thoughts?

AdamRJensen, 2024-03-15:
> I suggest we change the colors, though. To me, it would make sense as follows:
> 
> * active: green
> * inactive: red
> * unknown: blue or grey
> 
> Thoughts?

I agree. Let's make unknown blue.

For @kandersolar, the time period for active stations ends with `-`, unknown stations with a  `?`, and inactive with an end year.

PR #189 (2024-04-03): Update map legend markers to match markers on the map
Closes #96, using part of #117.

Instead of using font-awesome symbols (the existence of which varies across font-awesome versions, as noted in #96), this copies how the map is currently displaying the symbols (SVG).
closed 2024-04-03 by AdamRJensen