AssessingSolar/solarstations #111: Metadata formatting

AdamRJensen, 2024-02-04:
I've opened this PR to discuss how the metadata should be specified.

Specifically, the three columns: Tier, Instrument, Components. Right now, the Instrument column denotes either "Thermopile" or SPN1/RSR, and the the thermopile components measured are specified in the components columns (e.g., G;B;D). This is somewhat inconsistent as SPN1 also uses thermopiles... 

Perhaps these two columns could be combined into one, e.g., a station with an RSR and a traditional unshaded pyranometer (GHI) and pyrheliometer (DNI) would have an entry in the Instrumentation column of `G;B;RSR`.

Also, given the definition of Tier 1 vs 2, it should be possible to determine if a station is Tier 1 by checking if `G;B;D` is listed in the components column. So perhaps the Tier column should be derived and not be hardcoded. Thoughts?

AdamRJensen, 2024-02-05:
As per @IoannisSifnaios's comment:
> I think it is a good idea to add a third status called "unknown". E.g., I could not find info if the Saudi Arabian or the Chilean network is still in operation, and by default they would be categorized as "active". However, that might require going over the stations again to make sure that they are in the correct category...

I think that is a good idea. Perhaps we can use a question mark? E.g., ``2012-2014&2018-?`` Thoughts @kandersolar?

kandersolar, 2024-02-05:
An alternative approach: indicate instrumentation type and component presence/absence in a combined form.  Something like this:

| Station | DNI           | DHI                     | GHI        |
|---------|---------------|-------------------------|------------|
| Stn 1   | pyrheliometer | shadowball + thermopile | thermopile |
| Stn 2   | N/A           | SPN1                    | SPN1       |

kandersolar, 2024-02-06:
Question mark indicating uncertain end date makes sense to me!

PR #231 (2024-10-18): Change to instrumentation column
Closes #111 (required implementation of #230 and #229)
Closes #182

The following code was run to make the modifications:

    df = pd.read_csv('../solarstations.csv', dtype=str)
    df= df.drop(columns=['Tier', 'Instrument'])
    df= df.rename(columns={'Components': 'Instrumentation'})
    df.to_csv('../solarstations.csv', index=False)
closed 2024-10-18 by AdamRJensen