AssessingSolar/solarstations #140: Add stations from the Indian Meteorological Department (IMD)

IoannisSifnaios, 2024-02-19:
Stations described [here](chrome-extension://efaidnbmnnnibpcajpcglclefindmkaj/https://imdpune.gov.in/library/public/Solar%20Radiant%20Energy%20Over%20India.pdf).

AdamRJensen, 2024-02-20:
It seems that India put out a tender in 2019 for new pyrheliometers and trackers: https://internal.imd.gov.in/tenders/20191009_td_224.pdf

There's a list of stations on page 35 of the above document mention 22 stations.

There's another tender document from 2023 that only lists 18 stations (page 30): https://internal.imd.gov.in/tenders/20230211_td_296.pdf

Many of the stations added in #141 are on the list, which means that most likely the majority are still in operation.

AdamRJensen, 2024-02-20:
@IoannisSifnaios do you think it's the same stations as the Indian SRRA network described here (51 stations measuring GHI, DHI, DNI): https://www.researchgate.net/publication/275533998_Development_and_Test_of_Gap_Filling_Procedures_for_Solar_Radiation_Data_of_the_Indian_SRRA_Measurement_Network

IoannisSifnaios, 2024-02-21:
Pune is the only Tier 1 station ([photo](https://www.imdpune.gov.in/SI/radiation_lab_new.html))

PR #141 (2024-03-08): Add IMD (India) stations
closes #140
closed 2024-03-08 by AdamRJensen