AssessingSolar/solarstations #142: Check all hyperlinks

AdamRJensen, 2024-02-20:
We should write a script that checks that all the hyperlinks work.

kandersolar, 2024-02-20:
This could be done in the sphinx build with something like [sphinx.builders.linkcheck.CheckExternalLinksBuilder](https://www.sphinx-doc.org/en/master/usage/builders/index.html#sphinx.builders.linkcheck.CheckExternalLinksBuilder), or as a standalone CI job just using `requests`.  I'm inclined towards the latter so as to not slow down the documentation build.  @AdamRJensen any preference?

AdamRJensen, 2024-02-20:
The latter sounds great! Didn't Echedey do something similar to this in pvlib recently?

kandersolar, 2024-02-20:
Yes, but taking the sphinx approach: https://github.com/pvlib/pvlib-python/pull/1794

It makes more sense there since sphinx knows how to hunt down all the URLs in docstrings and such.  But here they are already in a nice tidy table for us :)

echedey-ls, 2024-03-01:
I just came across this. I recommend the sphinx builder approach if you have intra-links since they also would get checked.

PR #143 (2024-03-15): Add CI job for validating CSV entries
Closes #142 and #166.

#142 is about validating the hyperlinks in the CSV file, but it is possible and perhaps useful to extend the validation to other fields as well.  As an example, I have added basic sanity checks on the latitude, longitude, and elevation columns.

The job also adds annotations/comments in the PR on lines that fail the validation checks.
closed 2024-03-15 by AdamRJensen