AssessingSolar/solarstations #166: Implement data field checks

AdamRJensen, 2024-03-15:
Country names should be in this list (with some exceptions, e.g., Reunion): https://en.wikipedia.org/wiki/List_of_alternative_country_names

- Latitude should be within +/- 90 degrees
- Longitude should be within +/-180 degrees
- Acronyms should not be duplicated
- Elevation should be within 0 and 8800 m
- Data availability should be either: ``
- Tier should either be 1, 2 or ?
- Instrument type should be in list: [Thermopile, RSI, SPN1, RSR]
- Components should be in [G,B,D,IR,UVA, UVB]

This will help ensure consistent naming and acts as a pre-review and save reviewing time

kandersolar, 2024-03-15:
#143 includes some of these checks (lat, lon, elevation) and is designed in such a way to make it easy to add additional checks

IoannisSifnaios, 2024-06-14:
I think the options for data availability should be: `Freely`, `Upon request`, and `Not available`.

Leaving it `blank`/`None` just makes it more confusing...

PR #232 (2024-10-21): Add check_instrumentation to validate-csv.py
Closes #166
closed 2024-10-19 by AdamRJensen