AssessingSolar/solarstations #190: Add INES station in France

IoannisSifnaios, 2024-04-02:
Le Bourget-du-Lac, France, 45.64 N, 5.87 E, 235m, Tier 2, G;B

Information [here](https://pdf.sciencedirectassets.com/271089/1-s2.0-S0378778821X00229/1-s2.0-S0378778821009063/main.pdf?X-Amz-Security-Token=IQoJb3JpZ2luX2VjEDEaCXVzLWVhc3QtMSJHMEUCIQC4OHi5NwPUJLDNzowptSPff%2FCh6G44gmoCDEkA1js%2BlQIgTsmqObmhYENfoF0VIrXcnIw5tcl7mcsTE1pdZfcYzR0qswUIWRAFGgwwNTkwMDM1NDY4NjUiDFHwr928Nvz0qsM%2BsCqQBTwwK5viqd3u0AonaUg0wFjul6VB6qLEPjsOaYMVC69X%2FUhNhWIPQkV01D0A4fE9DAT2fe61AuG3FH0g8C70xQx5F8bqSxhxOZMa8czSop4jRFwGKTqShiciE92A%2BbrITJMb%2Fg78EnRmZrbClU1InulM0BgW5F0C24arkILp88AFdFgC5YcgSYrHX5T744mhWWeT2ZAvbd%2BpcF2sGXMn5UCa4zEa0ovAPfQ5q7fLhQLD6343Evfu0aFkl2BbBwMVhXsLlvGL%2FuxiNCElAV8nkVAtdeD0CI5dqZQJe6vm4Qfbp5VyDQr%2FLBJoR%2BaVbOZP%2FGuSDVTh7vnixg3NpSRuPSngZdbC2av1Mke72wT5Lj8%2BKzICWN1ElQykZYhNedICKmqIq2Vp1AqerRfvyetdEPl7dmYwWqo1bnXEhp%2F7dAsXBwvA0cy1xtmszqALO7BJ4D3rd0kEvKY6niuT%2BRvYd0vE4J%2F0WxtS%2FOnnAkB5x8FnldhRecTxVS0IgTdCeRn8sLSm4b626aXmmFHztUzNt5PwIHIszQPQ%2FpjpFPKvtuVJj4bvw1fqVLMvo4i5WJVzl9lS6VYiEUChHnPqsAxjzTBzqsvPXtOVtLbUmXR9GYj2IL8tLKlDbfY9hXPlA7lTDRwDA3ymq5UfhKk%2FFdUayJ6k6ttzmqbb9raoqBPo8ynuCewSi%2FyfE8fTQ0%2Fu88FGFoQgfTtKefB%2BduWP3M2RbqbM0mV%2Br4NW0wiiSv9dSqltF5xYzuXSuOyRlEdEIlhH14IC1SHCzHx%2BKd2tkUGjLJACSjZEp6MoSA3chluipz485BziCIbxDKN%2FtNfXAsfqjDw%2B9vSR074UznEKPT4belyhpzFutesL8GFsb7HpVZ%2FdMND%2BrrAGOrEBijmfhKmsqP1NpBNiCp%2FFb6gx5767m7331xtaUkglvgbRrgaSv%2FPc3fmhwTFNH7Qab7tXlEmuFwi3jZ1AI00lpdBHe8NH%2Bkfg1InkMPi%2BkB%2FgMQoMHUZI%2Fm0OLgayhETvbv4aR2TRxJ4xkrQCDbCIcRsH0I3P2S1ZTrdPCJMGuRaBBIZrL5w%2BdRoudR80auSQtBCGM%2FITadtZVTfGzuGF88Mw0aQVbuui6ZsGwe8rl8kO&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Date=20240402T093614Z&X-Amz-SignedHeaders=host&X-Amz-Expires=300&X-Amz-Credential=ASIAQ3PHCVTYV3ZBW2SS%2F20240402%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Signature=cdbb733cfa9826894cd50726dd78ae1bfe0951788d1d19e2f6e2587188591c63&hash=6752d835dda5adb23d6e6f5345aa867988c3887fc78c212b58a6ca51f4f23cf1&host=68042c943591013ac2b2430a89b270f6af2c76d8dfd086a07176afe7c76c2c61&pii=S0378778821009063&tid=spdf-fbbab312-92e0-4a1a-8408-eb70963d582c&sid=da45d4342a065142566987661a625744d4aagxrqa&type=client&tsoh=d3d3LnNjaWVuY2VkaXJlY3QuY29t&ua=1e0d5d500304035b5159&rr=86dfd559ed3eabd2&cc=dk).

@AdamRJensen do you have photos of the station?

AdamRJensen, 2024-04-07:
Here's a photo

[image]

Location: 45.641676, 5.875256

IoannisSifnaios, 2024-04-08:
It is not clear when their station started operating. INES was founded in 2005, they say on their [website ](https://www.ines-solaire.org/en/about-ines/)that by 2013, they had inaugurated all their research buildings, but in [this study](https://www.sciencedirect.com/science/article/pii/S0378778821009063?fr=RR-2&ref=pdf_download&rr=87109afdf9d58f57#b0390), data was used for 2019.

So I guess if we say 2019 we are definitely safe, but it is possible to be running for longer. Thoughts?

closed 2024-04-15 by AdamRJensen