AssessingSolar/solarstations #217: News and missings stations from the Indian Ocean Solar Network (IOS-net)

bilbaoba, 2024-07-02:
Hi Adam, 

We had a few more stations last two years in the IOS-net and I see also some are missing.

There is the list : 

```txt
             alias                                               name  territory altitude  latitude longitude                     doi             minDate             maxDate
4           aurere                                             Aurere La Réunion   603.00 -21.01977  55.42472 10.5281/zenodo.10693606 2023-10-30 12:05:00 2024-07-02 02:37:00
13 edfboisdenefles                         Saint Denis Bois de Nefles La Réunion   337.00 -20.91731  55.47644 10.5281/zenodo.10693525 2012-10-11 06:31:00 2023-02-20 04:19:00
14 edflapossession                                 La Possession Baie La Réunion    13.00 -20.93061  55.32897 10.5281/zenodo.10693552 2012-10-11 06:36:00 2023-05-07 09:26:00
15   edfsaintandre                             Saint Andre Chevrettes La Réunion   200.00 -20.96280  55.62243 10.5281/zenodo.10693536 2012-10-11 06:34:00 2023-02-20 05:30:00
16     edfsaintleu                                   Saint Leu Mazeau La Réunion   232.00 -21.20064  55.30233 10.5281/zenodo.10693559 2012-10-11 07:30:00 2024-07-02 02:38:00
17  edfsaintpierre                              Saint Pierre La Valle La Réunion    67.00 -21.31392  55.45107 10.5281/zenodo.10693567 2012-10-11 06:33:00 2024-07-02 02:39:00
21           marla                                              Marla La Réunion  1593.25 -21.10237  55.43058  10.5281/zenodo.8419976 2023-06-16 07:54:00 2024-07-01 16:06:00
33  saintpaulcarat                                Saint Paul Le Carat La Réunion   100.00 -21.04323  55.25089  10.5281/zenodo.8419974 2022-10-24 14:43:00 2024-07-02 02:40:00
34       uomreduit                     University of Mauritius Reduit  Mauritius   200.00 -20.24497  57.48770 10.5281/zenodo.10693484 2022-12-05 13:58:00 2023-04-25 09:13:00
```

Aurère and Marla are stations into the caldera of Mafate in La Réunion (check google maps). 
I should also add three more stations to Reunion by the end of the year, including one on the volcano.

How do you prefers we proceed ?  I can do a **PR** for you if needed. 

I remark in the CSV file that "Components" is missing for all our stations (CSV file inconsistent).

```error
In fread(ss_url, fill = F, header = T, sep = ",") :
  Stopped early on line 195. Expected 16 fields but found 15. 
```

It is mostly Glo and Dir measures on all the stations. 
URL can be https://galilee.univ-reunion.fr/ for all stations. 

Thanks

AdamRJensen, 2024-07-10:
@bilbaoba a PR would be awesome!

For the components, you can list the following:
- G for a pyranometer measuring GHI
- D for a pyranometer shaded by a tracking shading ball
- B for a pyrheliometer
- RSR
- SPN1

IoannisSifnaios, 2024-10-22:
I added the stations mentioned above. Based on the IOS-net website, I assumed that all of them have an SPN1 and UV instruments.

PR #246 (2024-10-23): add ios-net stations
closes #217
closed 2024-10-23 by AdamRJensen