AssessingSolar/solarstations #236: Add Finish FMI stations

AdamRJensen, 2024-10-21:
https://en.ilmatieteenlaitos.fi/weather-observations

IoannisSifnaios, 2024-10-21:
@AdamRJensen, has somebody told you that they are actually doing DNI and DHI measurements? On their website, they give very little information, and when trying to download solar irradiance data, they only have GHI, LWD, and UV. 

[image]

I am not convinced...

AdamRJensen, 2024-10-22:
I have sent an email to contacts at the University of Turku.

IoannisSifnaios, 2024-10-22:
Based on the email from Turku, there are three Tier 1 active stations in 2024. Data can be accessed [here](https://en.ilmatieteenlaitos.fi/download-observations).

Just for documentation:
* The website is not very user-friendly in identifying historical measurements. When I was trying to figure out when the Tier 1 stations started measuring DNI, more stations were popping up for some years, so it is possible that historical DNI data are available for more locations.
* Also, there are a number of stations measuring only GHI and DHI, but since they don't give specific information on the instruments used, I did not include them.

AdamRJensen, 2024-10-23:
One of the stations is duplicate: the space station one. I suggest consulting the two websites to see how best to merge the info. I suggest keeping the comment about poor data quality.

PR #245 (2024-10-23): Add FMI stations
closes #236
closed 2024-10-23 by AdamRJensen