AssessingSolar/solarstations #278: Gueymard stations part 3

IoannisSifnaios, 2024-11-15:
- [ ] Ouarzazate,,,Morocco, MASEN
- [ ] Cedar City, USA (UT), 37.696, -113.165, 1675, 2010-2013, NREL
- [ ] Elizabeth City, USA (NC), 36.282, -76.216, 26, 2010-2014, NREL
- [ ] Escalante, USA (NM), 35.4186, -108.08828, 2106, 2010-2013, NREL
- [ ] Humboldt, USA (CA), 40.876, -124.08, 36, 2015-2020, NREL
- [ ] Lafayette, USA (LA), 30.2051, -92.3979, 5, 2018-2021, NREL
- [ ] Lanai, USA (HI), 20.76685, -156.92291, 381, 2010-2012, NREL
- [ ] Loyola, USA (CA), 33.9667, -118.4226, 27, 2015-2016, NREL
- [ ] Milford, USA (UT), 38.4143, -113.0297, 1550, 2010-2013, NREL
- [ ] ORNL, USA (TN), 35.93, -84.3095, 245, 2015-2019, NREL
- [ ] Phoenix, USA (CO), 33.4166, -112.0188, 347, 2010-2013, NREL
- [ ] San Luis Valley, USA (CO), 37.561, -106.0864, 2335, 2008-2010, NREL
- [ ] Swink, USA (CO), 38.01221, -103.61696, 1255, 2011-2015, NREL
- [ ] Tucson, USA (AZ), 32.2297, -110.9553, 786, 2015-2019, NREL
- [ ] Watkins, USA (CO), 39.75685, -104.62025, 1674, 2013-2020, NREL
- [ ] Antofagasta, Chile, -24.09, -69.929, 965, 2017-2019, PSDA
- [ ] Padova, Italy, 45.4, 11.9, 2011-2012, University of Padova
- [ ] Bolzano, Italy, 46.455, 11.329, 250, University of Rome
- [ ] Rome Tor Vergata, Italy, 41.18556, 12.6233, 100, 2008-2017, University of Rome
- [ ] Chilbolton, United Kingdom, 51.145101, -1.439508, 82, 2020-2024, University of Southampton
- [ ] Timisoara, Romania, 45.7471, 21.2306, 87, 2009-2012, UVT, http://solar.physics.uvt.ro/srms/
- [ ] Bouzareah, Algeria, 36.79, 3.17, 345
- [ ] El Goléa, Algeria, 30.571, 2.87, 380

Note that the dates are the dates thaat Gueymard has data for, so the station can be active before/after.

AdamRJensen, 2025-03-14:
Wrote Chris about the Algeria stations.

AdamRJensen, 2025-03-17:
Emailed contact person in Algeria.