AssessingSolar/solarstations #282: NREL MIDC inactive stations

AdamRJensen, 2024-11-25:
- [ ] Cedar City, USA (UT), 37.696, -113.165, 1675, 2010-2013, NREL
- [ ] Elizabeth City, USA (NC), 36.282, -76.216, 26, 2010-2014, NREL
- [ ] Escalante, USA (NM), 35.4186, -108.08828, 2106, 2010-2013, NREL
- [ ] Humboldt, USA (CA), 40.876, -124.08, 36, 2015-2020, NREL
- [ ] Lafayette, USA (LA), 30.2051, -92.3979, 5, 2018-2021, NREL
- [ ] Lanai, USA (HI), 20.76685, -156.92291, 381, 2010-2012, NREL
- [ ] Loyola, USA (CA), 33.9667, -118.4226, 27, 2015-2016, NREL
- [ ] Milford, USA (UT), 38.4143, -113.0297, 1550, 2010-2013, NREL
- [ ] ORNL, USA (TN), 35.93, -84.3095, 245, 2015-2019, NREL
- [ ] Phoenix, USA (CO), 33.4166, -112.0188, 347, 2010-2013, NREL
- [ ] San Luis Valley, USA (CO), 37.561, -106.0864, 2335, 2008-2010, NREL
- [ ] Swink, USA (CO), 38.01221, -103.61696, 1255, 2011-2015, NREL
- [ ] Tucson, USA (AZ), 32.2297, -110.9553, 786, 2015-2019, NREL
- [ ] Watkins, USA (CO), 39.75685, -104.62025, 1674, 2013-2020, NREL

AdamRJensen, 2024-11-25:
@IoannisSifnaios

AdamRJensen, 2024-11-26:
Phoenix is called SOLRMPA SSSCAT: https://midcdmz.nrel.gov/apps/sitehome.pl?site=SSSCAT
Tucson is called SOLRMAP OASIS: https://midcdmz.nrel.gov/apps/sitehome.pl?site=UAT
Watkins is called [SOLARTAC](https://midcdmz.nrel.gov/apps/sitehome.pl?site=STAC) which is already included

IoannisSifnaios, 2024-11-27:
Additional inactive stations added:

- [ ] [ARM Radiometer Characterization System (RCS)](https://midcdmz.nrel.gov/apps/sitehome.pl?site=RCS)
- [ ] [Bluefield State College](https://midcdmz.nrel.gov/apps/sitehome.pl?site=BS)
- [ ] [SOLRMAP Kalaeloa Oahu (RSR)](https://midcdmz.nrel.gov/apps/sitehome.pl?site=OAHURS1M)
- [ ] [Lowry Range Solar Station (RSR)](https://midcdmz.nrel.gov/apps/sitehome.pl?site=LRSS)
- [ ] [San Clemente Island Data](https://midcdmz.nrel.gov/apps/sitehome.pl?site=SCI)
- [ ] [Sacramento Municipal Utility District (Anatolia)](https://midcdmz.nrel.gov/apps/sitehome.pl?site=SMUDA)
- [ ] [South Park Mountain Data](https://midcdmz.nrel.gov/apps/sitehome.pl?site=SPMD)
- [ ] [US Virgin Islands St. Thomas Bovoni and US Virgin Islands St. Croix Longford](https://midcdmz.nrel.gov/apps/sitehome.pl?site=USVIBOVB)
- [ ] [NREL Vehicle Testing and Integration Facility RSR](https://midcdmz.nrel.gov/apps/sitehome.pl?site=VTIF)
- [ ] [Xcel Energy Comanche Station](https://midcdmz.nrel.gov/apps/sitehome.pl?site=XECS)

PR #283 (2024-11-27): NREL MIDC inactive stations
closes #282
closed 2024-11-27 by AdamRJensen