AssessingSolar/solarstations #316: Add stations in Israel

AdamRJensen, 2025-05-08:
Israel has a number of multi-component irradiance monitoring stations. The data is available from: https://ims.gov.il/en/data_gov

These are stations measuring GHI
[image]

When selecting hourly radiation, the following stations are shown:

[image]

Based on info from this [link](https://ims.gov.il/en/node/175), then direct irradiance with a pyrheliometer and diffuse irradiance is measured with a shadow band at the Bet-Dagan site.

This article https://doi.org/10.1016/j.enconman.2015.03.077 also mentions DNI measurements at the Beer Sheva site from 1985 to 2013.
> The present analysis utilizes measurements recorded by the Solar Energy Laboratory’s meteorological station during the time interval October 1985 through August 2013. The meteorological station is located in Beer Sheva on the main campus of the Ben-Gurion University of the Negev (31°15′N, 34°45′E, 315 m m.s.l.), i.e., on the roof of the Poster Building. Beer Sheva is situated at the northern edge of the semi-arid Negev region of Israel. The global irradiation, on both a horizontal surface and a surface tilted at 40° to the south, is measured using Eppley Precision Spectral Pyranometers, Model PSP. The normal incidence beam irradiation is measured using an Eppley Normal Incidence Pyrheliometer, Model NIP. This station is part of the national network of meteorological stations and the Israel Meteorological Service checks the instruments’ calibration constants at regular intervals. A Campbell Scientific Instruments data-logger, Model CR10, monitors and stores the data at 10-min intervals (i.e., the meters are scanned at 10 s intervals and average values at 10 min intervals are calculated and stored). The data are transmitted periodically from the data-logger to the Solar Energy Laboratory via modem.

AdamRJensen, 2025-05-11:
Email from IMS
> ASHALIM – diffused + direct
YOTVATA – diffused only
BEER SHEVA – diffused only
BESOR FARM – diffused + direct
TECGNION HAIFA – diffused only
JERUSALEM GIVAT RAM – diffused + direct ( there are some environmental constraints )
BET DAGAN – diffused + direct
The equipment is :
Diffused – CMP11 ( KIPP & ZONEN )
Direct – sNIP ( Eppley )

AdamRJensen, 2025-05-11:
Ashalim,,,Israel,
Jesrusalem Givat Ram,,,Israel,1990,G;B;D
Besor Farm,,,Israel,2001-,G;B;D
Bet Dagan,,,Israel,1991-,G;B;D

Comment: unknown if DHI is measured with shadowband.

IoannisSifnaios, 2025-05-12:
I have the impression that all of these stations also have GHI measurements. So the stations you have omitted in your comment (i.e., Yotvata, Beer Sheva, Tecgnion Haifa) would be Tier 2. Don't we also want to include them?

AdamRJensen, 2025-05-12:
> I have the impression that all of these stations also have GHI measurements. So the stations you have omitted in your comment (i.e., Yotvata, Beer Sheva, Tecgnion Haifa) would be Tier 2. Don't we also want to include them?

I am pretty confident that the stations that only have GHI and DHI are using shadow bands for DHI, which would make them Tier 3 and not relevant for the catalog.

PR #317 (2025-05-13): Add IMS stations
Closes #316
closed 2025-05-13 by IoannisSifnaios